intel-aero/meta-intel-aero #400: intel aero flight controller bricked after flash px4 master

Huibean, 2019-01-29:
After flash px4 newest master, can not connect to Flight controller, can not flash again with aerofc-update.sh 
aerofc-force-bootloader-pin.py 1 tried
reinstall yocto iamge tried
cd /etc/fpga/
jam -aprogram aero-rtf-recovery tried
```shell
root@intel-aero:~# aero-get-version.py

BIOS_VERSION = Aero-01.00.16
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.2 (pyro)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc0
AeroFC firmware version = unknown
```

zehortigoza, 2019-01-30:
Try this:
- jam -aprogram aero-rtf-recovery.jam
- power down the board
- power on the board
- aerofc-update.sh <firmware>
- jam -aprogram aero-rtf.jam

lucasdemarchi, 2019-01-30:
Yep, no matter the state, what @zehortigoza said should solve it. Remember to power down the board (a reboot is not sufficient). And to reprogram the FPGA later.

lucasdemarchi, 2019-01-30:
(unless PX4 overwrote the bootloader... but that's very unlike)