intel-aero/meta-intel-aero #407: Installing IntelRealSense

PedroHRPBS, 2019-07-04:
I've been following the tutorials, I already got my intel aero flying with ubuntu installed.
But now I'm trying to install the intelrealsense SDK and this error is showing up.

intelaerodrone@intelaerodrone-CherryTrail:~/librealsense/build$ make
Scanning dependencies of target realsense
[  1%] Building CXX object CMakeFiles/realsense.dir/src/archive.cpp.o
[  2%] Building CXX object CMakeFiles/realsense.dir/src/context.cpp.o
[  3%] Building CXX object CMakeFiles/realsense.dir/src/device.cpp.o
[  4%] Building CXX object CMakeFiles/realsense.dir/src/ds-device.cpp.o
[  5%] Building CXX object CMakeFiles/realsense.dir/src/ds-private.cpp.o
[  6%] Building CXX object CMakeFiles/realsense.dir/src/f200.cpp.o
[  7%] Building CXX object CMakeFiles/realsense.dir/src/hw-monitor.cpp.o
[  9%] Building CXX object CMakeFiles/realsense.dir/src/image.cpp.o
[ 10%] Building CXX object CMakeFiles/realsense.dir/src/ivcam-private.cpp.o
[ 11%] Building CXX object CMakeFiles/realsense.dir/src/ivcam-device.cpp.o
[ 12%] Building CXX object CMakeFiles/realsense.dir/src/log.cpp.o
[ 13%] Building CXX object CMakeFiles/realsense.dir/src/motion-module.cpp.o
[ 14%] Building CXX object CMakeFiles/realsense.dir/src/r200.cpp.o
[ 15%] Building CXX object CMakeFiles/realsense.dir/src/rs.cpp.o
[ 17%] Building CXX object CMakeFiles/realsense.dir/src/sr300.cpp.o
[ 18%] Building CXX object CMakeFiles/realsense.dir/src/stream.cpp.o
[ 19%] Building CXX object CMakeFiles/realsense.dir/src/sync.cpp.o
[ 20%] Building CXX object CMakeFiles/realsense.dir/src/timestamps.cpp.o
[ 21%] Building CXX object CMakeFiles/realsense.dir/src/types.cpp.o
[ 22%] Building CXX object CMakeFiles/realsense.dir/src/uvc-libuvc.cpp.o
[ 23%] Building CXX object CMakeFiles/realsense.dir/src/uvc-v4l2.cpp.o
/home/intelaerodrone/librealsense/src/uvc-v4l2.cpp: In constructor ‘rsimpl::uvc::subdevice::subdevice(const string&)’:
/home/intelaerodrone/librealsense/src/uvc-v4l2.cpp:158:39: error: ‘V4L2_CAP_META_CAPTURE’ was not declared in this scope
                 if((cap.device_caps & V4L2_CAP_META_CAPTURE)){
                                       ^
CMakeFiles/realsense.dir/build.make:542: recipe for target 'CMakeFiles/realsense.dir/src/uvc-v4l2.cpp.o' failed
make[2]: *** [CMakeFiles/realsense.dir/src/uvc-v4l2.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/realsense.dir/all' failed
make[1]: *** [CMakeFiles/realsense.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

sizsJEon, 2019-07-08:
If you are using ROS. there is a solution.

before make libsense library, 
install ros first.
https://github.com/intel-aero/meta-intel-aero/wiki/05-Autonomous-drone-programming-with-ROS

it works for me.

houattas, 2019-07-12:
Hi,
I got the same problem.
Did it work for you? @P or is there another solution because the one proposed did not work for me

PedroHRPBS, 2019-07-12:
@houattas Hi!
Actually, to solve it I had to undo the last committed changes. Because this bug was added on the last PR as I could figure out.
So I deleted the added lines of code that appears on the diff between the last comit and the previous.

houattas, 2019-07-12:
Thanks it works :)