intel-aero/meta-intel-aero #410: Issue with Depth calculation of Intel Realsense camera

RohanKurane, 2019-07-09:
Hello, 
I am having an issue with the depth calculation of the intel realsense camera frame. It always seems to return me a 0. 
Here is how I get it after receiving a frame from `rs_wait_for_frames()`
`
uint16_t *depth = ( uint16_t* )   rs_get_frame_data( camera, RS_STREAM_DEPTH, &e);
`
I then use it to calculate the rgb values as
```
               for( int i = 0, end = 640 * 480; i < end; ++i )
               {
                   //rgbdata[ i ].r = depth[ i ] * 60 / 999;
                   rgbdata[ i ].r /= 4;
                   rgbdata[ i ].g /= 4;
                   rgbdata[ i ].b /= 4;
               }
```

What am I doing wrong ? 

@lbegani @lucasdemarchi

RohanKurane, 2019-07-09:
Im not sure if this is a depth issue or what. The video appears all dark. How can I get more color ?

RohanKurane, 2019-07-09:
I think its an issue with the RGB intensity values ? Should I use the RGB values for each frame as they are or should I process them like above divides each by 4 ?

RohanKurane, 2019-07-11:
[image]

Here is a pic how the video appears like on QGC. 
@lbegani @lucasdemarchi @lfelipe @anselmolsm  Any idea what setting of the camera needs to be changed ?

lfelipe, 2019-07-11:
Hey @RohanKurane . Unfortunately not sure exactly what might be the problem you're facing here.

This particular repository here is not being maintained anymore as the product has been discontinued. I saw you already posted on the Intel Aero Forum, but would suggest possibly asking this particular question on the Realsense forums to see if anyone there might be able to help you out (as it doesn't seem to be particularly drone specific and more related with properly obtaining/interpreting data from the realsense camera itself).

RohanKurane, 2019-07-11:
Thanks @lfelipe 
I will do that. Yea,this seems odd.

RohanKurane, 2019-07-11:
hey @lfelipe just wanted to give an update - I tried this outside and the video worked just fine. do you know what setting on the realsense camera I need to adjust for it to work in less light or the dark as well ?

RohanKurane, 2019-07-11:
think it might be the Infrared. But I cant seem to turn it off. Tried disabling the infrared stream. Also disabling the Emitter.

closed 2019-07-17 by RohanKurane