intel-aero/meta-intel-aero #415: No GUIDED mode

EstebanGameDevelopment, 2019-11-12:
Hello,

I'm trying to develop with Python with DroneKit. The examples that I work with always ask to set up the mode to GUIDED, but with Intel Areo is nowhere to be seen. Am I missing any configuration in order for this option to appear or it's no longer an available mode?

[MavLink: Image where I show how there is no GUIDED mode available](http://www.yourvrexperience.com/WHY_GUIDED_IS_NOT_AVAILABLE.png)

Any help would be appreciated.

Thanks.

lucasdemarchi, 2019-11-12:
In ardupilot it's called guided mode. In px4 it's offboard

EstebanGameDevelopment, 2019-11-14:
> In ardupilot it's called guided mode. In px4 it's offboard

Thanks for the reply. I seem to be able to change the mode with DroneKit by:

`from dronekit import connect, VehicleMode, LocationGlobalRelative`
`import time`
`from pymavlink import mavutil`
`vehicle = connect('tcp:127.0.0.1:5760', wait_ready=False)`
`vehicle.mode = VehicleMode("OFFBOARD")`

Next, I'm able to arm the drone:

`vehicle.armed = True`

But when I send the command:

`vehicle.simple_takeoff(aTargetAltitude)`

it's ignored and I don't know why.

Also another thing that worries me even more is that when I try to use MAVProxy commands in order to perform operations of take off in OFFBOARD mode there is no way that the drone accepts the command:

[MAVProxy try to change the mode to OFFBOARD does not work](http://www.yourvrexperience.com/mode_offboard.png)

Does anyone have made MAVProxy commands to change mode to OFFBOARD? 

**arm throttle** works but that is useless because I need change the mode.

EstebanGameDevelopment, 2019-11-15:
Could be possible that any of you could share a python DroneKit script that really works on Intel Aero? So far I have not been successful to fully perform any operation aside arming the drone.